AAPS-Omnipod/AndroidAPS-OLD #19: Add warning for uncertain bolus

bartsopers, 2019-12-07:
On rare occasions, we are not certain whether or not a bolus command has been successfully executed by the pod. In that case, we should show a warning and encourage the user to manually verify whether or not the bolus is being delivered by listening for clicks from the pod.

There is a TODO for showing the warning in `AapsOmnipodManager`:
`// TODO notify user about uncertain failure ---> we're unsure whether or not the bolus has been delivered`

andyrozman, 2019-12-08:
We could display warning in AAPS for that?

bartsopers, 2019-12-08:
It's important that the user sees the warning immediately, so they can check if they hear clicks from the pod. So the warning should show when the bolus progress dialog is still active.

Is that the case for an AAPS warning?

bartsopers, 2019-12-18:
The warning is now shown in the background and we don't use a sound, as the alert is not dismissable while the bolus progress dialog is showing.

Good enough for a first test version, but requires some future improvements. Will add an issue for that

closed 2019-12-18 by bartsopers