AAPS-Omnipod/AndroidAPS-OLD #22: add acknowledgement message of profile switch from pod

ZandMarc, 2019-12-14:
With AAPS with Combo, I get a message a few minutes after a profile switch, acknowledging that the profile in the pump has been updated. It would be nice if a similar message came from pod.

bartsopers, 2019-12-14:
Thanks for reporting, we’ll look into this. Can you please add an example screenshot of the message you get with the combo?

bartsopers, 2019-12-15:
Another possible solution for this would be to distinguish between basal and temp basal in beep configuration. 

You would always be notified when setting a basal schedule fails, so if you don't see an error message, you should know that it succeeded. You actually also rely on that for temp basals, as in general, these would be automatically set by AAPS, without any acknowledgement message to the user. 

If however you would like some extra confirmation specifically for basals, you could just enable the basal beeps (when we make a separate setting for temp basal beeps, which you could then disable).

ZandMarc, 2019-12-17:
[image]
[image]

andyrozman, 2019-12-22:
Done. I added both notifications, if profile change is OK and if profile change fails...

bartsopers, 2019-12-24:
@andyrozman these messages would also be very nice to have for the setTime command

ZandMarc, 2019-12-26:
Did a profile switch tonight and got the notifications. Perfect! Thx, @andyrozman !

closed 2019-12-26 by ZandMarc