AAPS-Omnipod/AndroidAPS-OLD #28: Field "serial number of pump" in RL stats tab has no value

ZandMarc, 2019-12-17:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

andyrozman, 2019-12-22:
Are you sure? If you Inited pod correctly, then address of Pod should be displayed...

ZandMarc, 2019-12-22:
[image]

ZandMarc, 2019-12-22:
Pod and rl are working fine. Still, address is not displayed and neither is the serial number of the pod

vanelsberg, 2020-01-23:
Confirming!

ZandMarc, 2020-01-25:
Tested: works for me!

vanelsberg, 2020-01-26:
Commit 36128c40:
Tested: works for me!

bartsopers, 2020-01-26:
As @ZandMarc and @vanelsberg  already noted: fixed by @andyrozman in 36128c4007d99aa882e821e30b94d9400fd3f67f

closed 2020-01-26 by bartsopers