AAPS-Omnipod/AndroidAPS-OLD #46: Sometimes info in Omnipod tab "hangs"

ZandMarc, 2019-12-29:
Sometimes the Omnipod tab seems to "stall". See screenshot attached: the  bolus of 3.6U mentioned after pod status, was ordered and delivered an hour before the screen shot. After that, several smb's were done. The "bolus of 4.4U, 52 min ago" was actually delivered almost 2 hrs before the screenshot.
[image]

andyrozman, 2019-12-29:
Last bolus is meant for last "real" bolus. SMBs are ignored...

ZandMarc, 2019-12-29:
That's different from how Combo tab behaves: there the last bolus shows the SMB's. But either way, that's not the point of this issue. This issue is about the bolus not "disappearing" from Pod status and "appearing" at Last bolus.

andyrozman, 2019-12-30:
Why should it disapper? It shows last Bolus. SMBs are part of oRef1 and shouldn't be displayed here... I have sometimes last bolus there for 6 hours (since the last bolus)...

ZandMarc, 2019-12-30:
I think it should be at the "Last bolus: " line, not hanging for hours at the "Pod status: " line

andyrozman, 2019-12-30:
Oh you meant unser Pod Status? Mhm... That is queue, this core thing, there is nothing I can do about that, but that should go out as soon as command is finished... Does this happen often?

ZandMarc, 2019-12-30:
Under pod status is what I meant, yes. I haven't kept track of it, but I think it happens a couple of times a day. I'll watch and keep track of it the next couple of days.

ZandMarc, 2020-01-25:
I haven't seen this happen in the last days/weeks. I'll close the issue for now.

closed 2020-01-25 by ZandMarc