AAPS-Omnipod/AndroidAPS-OLD #47: Text missing in "pair and prime" screen

ZandMarc, 2019-12-29:
In the "pair and prime" screen, some text is missing. Probably because the spinner and checkboxes cover it.
[image]

ZandMarc, 2019-12-29:
The same goes for "deactivating the pod" screen. Text is cut off after "please press Cancel and use"...

bartsopers, 2019-12-29:
I would personally (still) opt for removing the checkboxes and just show the spinner, as the checkboxes add no value for the user IMO. I feel that for the user, the only relevant thing is that the pairing and priming step succeeded, and not the individual steps like assigning an address. It might be useful for us devs, but we can already see that info in the logs.

Removing the checkboxes would save some space and possibly fix this issue.

andyrozman, 2019-12-29:
You don't even see the second checkbox, or is that just problem with taking snapshot?

ZandMarc, 2019-12-29:
I think the second checkbox is covered by the spinner (barely visible in the screenshot, but it's there). I second Barts suggestion to remove the checkboxes, since they don't add info to the user. They are either both unchecked or both checked, so they don't offer info on the progress of the process.

andyrozman, 2020-01-01:
I changed fonts and sizes a little, so this should be fixed... Please reopen if its not fixed...

andyrozman, 2020-01-25:
Is this fixed? @ZandMarc

ZandMarc, 2020-01-25:
yes, this is fixed

closed 2020-01-25 by ZandMarc