AAPS-Omnipod/AndroidAPS-OLD #48: When Omnipod is selected for first time, no scan for RileyLink

ZandMarc, 2019-12-29:
When Omnipod is selected for the first time in the configurator, AAPS switches to the Overview tab (and inserts a tab for Omnipod). But as I understand it, after choosing Omnipod for the first time, it is necessary to scan for and select Rileylink. Therefore it would be logical to open the settings for Omnipod after it is selected.

andyrozman, 2019-12-30:
It doesn't work that way. You have to configure it Config Builder. When you select Omnipod, you have clog there and if you can click on it you can configure things. You don't have to select Omnipod go out and then select configure... That is the way AAPS works...

ZandMarc, 2019-12-30:
Also something for the docs then. New users start with the config builder, but there will also be users that have already used AAPS with another pump (or a virtual one) and will skip the config builder because they import their saved preferences. In the docs it would be nice to add a reminder for them that after changing the pump to Omnipod, they have to go back, select the cog and scan and select RL.

closed 2020-02-29 by andyrozman