AAPS-Omnipod/AndroidAPS-OLD #49: After scanning for and selecting Rileylink, it isn't initialized

ZandMarc, 2019-12-29:
After I have scanned for and selected an RL, AAPS doesn't initialize it. In the Omnipod tab it says "bluetooth connected" and the BT-symbol is spinning. Nothing further happens. After I forcefully stop AAPS and start it again, RL is initialized and connection to pod is made.
[image]

andyrozman, 2019-12-30:
Yes we have some problem with Medtronic... After you select it for first time, you have to restart application. I could never got this working, and after few tries I haven't continued with it. Problem is that sometimes it works and sometimes it doesn't...

ZandMarc, 2019-12-30:
Ok, something for the docs then.

bartsopers, 2020-08-18:
Can anyone confirm this issue still exists?

mistermintsgh, 2020-08-19:
> Can anyone confirm this issue still exists?

I get this issue (0.3)

vanelsberg, 2020-08-19:
Not sure.
On one of @bartsopers early "trial commits" to find the cause of "loosing the Pod after restart", 
I experienced something similar after trying to switch to another RL.

Wil retest switching/initializing RL for the dev/0.4.1 build.

bartsopers, 2020-09-04:
Closing this as I haven't any reports of this behavior in omnipod_eros_dev. If you run into this issue again, please create a new issue.

closed 2020-09-04 by bartsopers