AAPS-Omnipod/AndroidAPS-OLD #51: Text in deactivate isn't completely visible

ZandMarc, 2020-01-05:
Adjusting the font fixed the issue of several texts in initpod windows being unreadable. Only the font in "deactivate" seems unchanged, resulting in the last line being slightly cut off on my small (4.6 inch) screen 
[image]

andyrozman, 2020-01-12:
I made text area little bit smaller when I did the font change, but I can try to make little bit bigger... I am testing with relatively small phone (Samsung A5 (v1) = 5.0' screen, so I didn't think it will be a problem... 

I will make a change and you can test it (on branch), next week.

andyrozman, 2020-01-19:
@ZandMarc Is this fixed?

ZandMarc, 2020-01-19:
No. Just aesthetics though: Text is completely readable, only tails of p and y are cut off.

bartsopers, 2020-01-19:
I also experience this on my Pixel 3a, which has a quite large screen. 
[image]

andyrozman, 2020-01-24:
Is this still the problem?

ZandMarc, 2020-01-24:
I updated apk Wednesday and will deactivate this weekend: I’ll check and let you know (and close the issue if it is)

ZandMarc, 2020-01-26:
Better, but now I see there's another line, with 1 word ("state"?) That's not readable: see screenshot.
[image]

andyrozman, 2020-02-15:
Should be fixed now...

closed 2020-03-14 by andyrozman