AAPS-Omnipod/AndroidAPS-OLD #54: restarting AAPS with active pod gives erroneous message "no active pod"

ZandMarc, 2020-01-14:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

andyrozman, 2020-01-24:
Please if this happens again, I need logs... Without logs and snapshots of the state, there is not much I can do...

bartsopers, 2020-01-25:
[AndroidAPS.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4112036/AndroidAPS.log)
This log is for both this issue and #15. Please note that because of #15 (and possibly also this issue), there is no closed loop as AAPS thinks the pump is suspended.
The log starts after rebooting AAPS. After RL is initialized, I get the pump suspended notification. Please refer to below screenshot for the time.
[image]

I went to sleep again after this, and when I woke up, the pump suspended notification was still there, and there also was a No pod active notification now. That notification appeared somewhere between the first and the last screenshot. The log covers all time between both screenshots.
[image]

ZandMarc, 2020-01-25:
This is the log of the moment I opened this issue for. At 11.23-11.24 there is a notification "no active pod" in the logs. After that there's a "refresh" (done by me) and the active pod is picked up again.
[AndroidAPS._no_active_pod_false.txt](https://github.com/AAPS-Omnipod/AndroidAPS/files/4112043/AndroidAPS._no_active_pod_false.txt)

andyrozman, 2020-02-15:
This might be fixed already with fixes on #15

bartsopers, 2020-08-17:
This has probably been fixed. Please re-report if the problem still exists after the 0.4 release.

closed 2020-08-17 by bartsopers