AAPS-Omnipod/AndroidAPS-OLD #55: Time and/or Timezone changed on the pump notification

Kywalh, 2020-01-19:
Twice a day at noon and midnight, the following  green notification appears: 

> Time and/or Timezone changed on the pump

[image]

Kywalh, 2020-01-19:
#
[AndroidAPS._2020-01-18_13-10-47_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4082455/AndroidAPS._2020-01-18_13-10-47_.1.zip)

andyrozman, 2020-01-23:
I added some logging here... I am testing this on separate application and I am not noticing any time changes... (someone said he got changes at least every 12 hours)... So far I haven't seen anything... 

There will be log in AAPS from now on, so we can follow this problems

Kywalh, 2020-01-23:
At what time did you commit ?
I just built the last commits at 14:36...

whynot2018, 2020-01-27:
I have today's late morning green "Time and/or Timezone change on pump" notification.
AAPS commit was Jan 22, 2020 , f5af60712520eedfe7fc382a9b06a64f439fc46e
[image]

vanelsberg, 2020-01-29:
AndroidAPS 2.6-omnipod-0.2
Build EROS_0.2 (release)

Unexpected Time and/or Timezone ....
Had seen this message only one or two times until updating to V0.2 release.
Now getting this message about once a day (around 00:00h?).
Logs included

[image]
[AndroidAPS.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4125688/AndroidAPS.zip)

vanelsberg, 2020-01-30:
And again....
About at the same time of the day.
I would think this my phone scheduled time sync with the network?

[AndroidAPS_LOG_1580338904238.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4131498/AndroidAPS_LOG_1580338904238.log.zip)
[image]

vanelsberg, 2020-01-31:
Additional remark to my previous post of 30/01

### Observation:
Starting with the V0.2 release build I am seeing "Time and/or Timezone change on pump" notification *daily and always at 00:00 AM*

Kaivish, 2020-02-24:
Got this timezone issue tonight arount 00.00.58.
Prior to that aaps restarted by itself aroint 23.36 or 23.56. Don't know how or why. Batt. Optimizations are off for aaps. This caused the pump suspended issue resulting in a bad hypo in the middle of the night for several hours. When pump was suspended for 4.5 hrs, on pod tab it showed last connected 4.5hrs ago, tapping refresh connected to pod and got out of pod suspended state and loop started again. 
I guess this issue should get higher priority and be fixed. 

[AndroidAPS_LOG_1582515940110.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4243055/AndroidAPS_LOG_1582515940110.log.zip)

Kaivish, 2020-02-28:
Uploading logs. Same time zone notifications. Pump not suspended but in logs, there is mentions of pump suspended. Not so good at understanding the logs but it all at beginning of the logs
[AndroidAPS_LOG_1582931921452.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4269592/AndroidAPS_LOG_1582931921452.log.zip)

andyrozman, 2020-02-29:
It seems that we have 2 issues here. One is time change on midnight, and one are other time changes... I fixed the problem 1, and added a lot more of debug for he second problem... I also added functionality to disable Time changes.

mick711, 2020-03-05:
I can upload m'y log files this Time change appen lot of time all days with 0.2

teleriddler, 2020-03-17:
Uploading logs shortly but got the timezone message yesterday.  

Around 6 pm but it was visiting my doctors office that has no cell service (thick walls in the basement of the building) but they do have wifi.  It came as I left the office which I am assuming was when my phone switched from wifi connection back to the cellular network, which I imagine would trigger the cell service checking and setting my phones time to be in sync with the cell tower/service time.   

Just an idea of what might be happening in this case.

vanelsberg, 2020-04-29:
> time change on midnight,
Confirming. Haven seen any "time change" alerts around midnight with V0.3. 
Tnx!

bartsopers, 2020-08-18:
@teleriddler @andyrozman @vanelsberg does this issue still exist?

vanelsberg, 2020-08-19:
Have not seen this for a long time (since repair in 0.3?)

bartsopers, 2020-08-24:
Closing this as it seems to be working properly now.

closed 2020-08-24 by bartsopers