AAPS-Omnipod/AndroidAPS-OLD #56: Pump unreachable alert although option off

Kywalh, 2020-01-19:
Having the Rileylink and the pod at 2 different sites at home, I have a pump unreachable alarm although the option is ticked off in the settings...

[image]
[image]

[AndroidAPS._2020-01-19_00-00-00_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4082506/AndroidAPS._2020-01-19_00-00-00_.1.zip)

bartsopers, 2020-01-19:
This is actually by design. We chose to have a fixed pump unreachable warning, because once you use your pod with AAPS, there is no other way to control your pod, so it's important that the pod is reachable.
In a future version, we will remove the option to disable alerts, because as you found out, that option doesn't have an effect for the Omnipod driver.
Furthermore, recent changes in dev changed the alert sound behavior. That should make it possible to disable sounds for notifications. We'll merge these changes in a future test version.

I'll close this issue and create a new issue for removing the option to disable alerts.

closed 2020-01-19 by bartsopers

bartsopers, 2020-01-19:
Opened #59 for this