AAPS-Omnipod/AndroidAPS-OLD #60: Priming a pod sometimes fails

bartsopers, 2020-01-19:
Although pairing & pairing works for most people, several users have reported failures in priming. It seems that assigning an address always succeeds, but priming doesn't. It seems to fail with a communication timeout.

We should investigate the cause of this issue. Loop and Omnicore user do not experience similar issues, so I suspect that there is a problem in our code somewhere.

oakeley, 2020-01-19:
[AndroidAPS._2020-01-19_11-10-12_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4082796/AndroidAPS._2020-01-19_11-10-12_.0.zip)

Will contain some Omnicore logs too

whynot2018, 2020-01-19:
I had the same issue.
Modified RileyLink 916MHz to BLE Bridge Complete Kit (RILEYLINK916)  October 2018

[AndroidAPS_old.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4082884/AndroidAPS_old.log)

Marije22, 2020-01-19:
First time pairing and priming a pod did not succeed for me. Tried several ways to make it work. Changing the pod seemed to be the only solution.  

[wetransfer-d45140.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4083097/wetransfer-d45140.zip)

vanelsberg, 2020-01-29:
See also issue "Priming a pod sometimes fails #60" ??

whynot2018, 2020-02-02:
That issue appeared again.
[AndroidAPS_2020-02-01_23-43-34.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4144280/AndroidAPS_2020-02-01_23-43-34.zip)

bartsopers, 2020-02-22:
[AndroidAPS_LOG_1580506295669.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4239037/AndroidAPS_LOG_1580506295669.log.zip)

avivaac, 2020-03-02:
Wasn't able to deactivate, or pair and prime a new pod this evening.  Tried everything, swapped RL and 2nd pod attempt and it eventually worked.  Ive attached logs and screenshots.  
[AndroidAPS.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4277470/AndroidAPS.log)
[AndroidAPS._2020-03-02_18-18-05_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4277471/AndroidAPS._2020-03-02_18-18-05_.0.zip)

avivaac, 2020-03-02:
[image]
[image]
[image]
[image]
[image]
[image]
[image]

nhobbs7, 2020-03-10:
I had a communication issue over the final hours of the pod (starting after it expired at 72 hours so I could not successfully communicate with the pod in the extra 8 hours). I could not deactivate the first pod so I selected "Reset Pod". When I initialized a new pod, the priming occurred successful and the canula insertion occurred properly, but there was a communication error and the app could not verify the insertion. The option to retry repeatedly failed to meet success on the insert canula screen and after leaving that window, when "Refresh" was selected on the Omnipod tab, the pod history message indicated "the pod is in a wrong state".  After several hours, when returning to the init pod option, it successfully connected.

AndroidAPS 2.6-Omnipod-0.2
Running on Moto X4 (US-Google Fi version). Android version 9.
-Traveled by Airplane (2 hour timezone shift) morning before issue occurred and did not verify the time updated correctly before sleeping multiple hours. Likely unrelated, but I can find the older logs if it may be useful.
-This may also fit under the "Communication Failed" issue as the cause may not relate to the priming itself.

[image]
[image]
[image]

[AndroidAPS_LOG_1583802929320.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4313946/AndroidAPS_LOG_1583802929320.log.zip)

bionictom, 2020-03-18:
Adding my log from the problem with priming.
It worked after restarting phone and RL

[image]

[AndroidAPS_LOG_1584568640610.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4351543/AndroidAPS_LOG_1584568640610.log.zip)

D-Cudz, 2020-04-19:
[AndroidAPS._2020-04-18_23-17-24_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4499763/AndroidAPS._2020-04-18_23-17-24_.0.zip)

Everything has been going well after trying this system for the past 2 weeks until now. I had a screamer when trying to pair the pod so threw that out, reset my phone and tried a new pod and then I got the error message seen in the attached log at about 10:50pm. AAPS said that I have to deactivate it and try a new one so I deactivated it on the app. Says it worked but then I checked to see if I could pair it with the Omnipod PDM and it paired fine so it couldn't have been deactivated. Strange. Thanks.

whynot2018, 2020-06-02:
Priming failure unsuccessful
[AndroidAPS._2020-06-02_17-44-24_.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4718968/AndroidAPS._2020-06-02_17-44-24_.zip)
[image]
[image]

whynot2018, 2020-06-09:
new priming issue
[AndroidAPS.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4754217/AndroidAPS.log)

bartsopers, 2020-08-17:
In 0.4 we did a lot of improvements in the Pod activation process. Closing this issue for now, please re-report if problems still exist after the 0.4 release

closed 2020-08-17 by bartsopers