AAPS-Omnipod/AndroidAPS-OLD #62: Expiration reminders don't match PDM

bartsopers, 2020-01-20:
The expiration reminders are copied from Loop. These are different than the PDM's, and I'm not sure why. Until we (might) make expiration reminders configurable in the future, I think it's more appropriate to match the PDM's behavior.

ZandMarc, 2020-01-21:
According to my 2014 manual, this is configurable in the PDM from 1 hour to 24 hrs before expiration. And Insulet takes 72 hrs as expiration. Default is set at 4 hours, i.e. 68 hrs, i.e. 12 hours before the real expiration (which is at 80 hrs). This is ok for me (though I would like a reminder a bit closer to the real expiration, but maybe that's something for a later date).

bartsopers, 2020-01-21:
Discussed this with @ZandMarc, and based on the Pod's normal lifetime of 72 hours, we think sensible values would be to 71 hours for the first alert (1 hour before you should change the Pod according to Insulet), and 79 hours for the seconds alert (1 hour before the Pod will die).

I also updated the nominal pod life to 72 hours (this was 77 hours), so the Pod info tab should now show an expiration date of 72 hours after the start date.

Fixed in 6f7a07185a62dae1f83b05c07451e788308d3f7f

closed 2020-01-21 by bartsopers