AAPS-Omnipod/AndroidAPS-OLD #70: Lock-up when selecting RL-Stats

vanelsberg, 2020-01-24:
AndroidAPS 2.6-omnipod0.2-SNAPSHOT
Build EROS_0.1-17-g22b8d0844-2020.01.24-18:31 fullrelease

# Problem
AAPS completely locks up after pressing the "RL-STATS" button with a completely back and non-responsive screen.
When waiting long I am presented the Wait/Close pup-up. Alternatively I can force-close through settings. From there AAPS can be restarted without issues.

* Additional remark:
This problem can be reproduces at will on both my Samsung S7 as the Note 9.
* Logfiles included:
Samsung S7: 
[AndroidAPS.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4110473/AndroidAPS.zip)

vanelsberg, 2020-01-26:
Problem appears to be solved with commit 36128c40

bartsopers, 2020-01-26:
This has been resolved as @vanelsberg mentioned

closed 2020-01-26 by bartsopers