AAPS-Omnipod/AndroidAPS-OLD #71: Bolus administred does not show in increments of 0.05

vanelsberg, 2020-01-26:
AndroidAPS 2.6-omnipod0.2-SNAPSHOT
Build EROS_0.1-18-g36128c400-2020.01.25 fullrelease

OmniPod bolus is in Nx0.05 units.
Om administering a bolus, progress bar also increments in unit of 0.05. On the POD tab (and also under POD history) boluses are only reported in increment of 0.1U.
(Basel rates are OK, showing in multiple unit of 0.05))

## Example:

*From overview, a manual bolus of 0.25 is given (pop-up confirms as "Bolus: 0.25U")
*On OK, the progress bar confirms delivering 0.25U
*POD tab reports: "Last bolus: 0,3 U (Now)"
*Pod History reports: "Set Bolus, 0.3 U, CH=0.0 g"

vanelsberg, 2020-02-01:
Found it: Looks like the 0.27U was the result of doing a profile switch to 110% while current profile rate was 0.25U

### Repro
While current basal rate in Profile is at 0.25U/h
Switch current profile to 110% 
POD tab shows 1.1 * 0.25 = 0.27 U/h

### Screenshot:
[image]

andyrozman, 2020-02-01:
This is fixed (will be released with 0.3)

closed 2020-03-14 by andyrozman