AAPS-Omnipod/AndroidAPS-OLD #78: Failed to activate pod with old v1 Eros settings

marksieminski, 2020-01-29:
Reporting bugs
--------------
- exported AAPS settings with connected pod to hard drive. Disconnected pod. Uninstalled Eros v1 and installed Eros v2.

Imported settings and noted pod address still available. Reset pod and tried to pair. Pod communication failed. Reset Riley link and reset phone, pod communication failed.

Uninstalled and reinstalled v2 with no settings imported. Connected Riley link and tried to pair pod. Pod failed.

- Eros version 2
- phone pixel 3 xl
- android 9
- failed to grab log.
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

marksieminski, 2020-01-29:
[image]

andyrozman, 2020-02-15:
Seems like one of... You should never export settings while Pod is running... Closing

closed 2020-02-15 by andyrozman