AAPS-Omnipod/AndroidAPS-OLD #79: Initial one-time failure at Pod Init / Pair & Prime

vanelsberg, 2020-01-29:
AndroidAPS 2.6-omnipod-0.2
Build EROS_0.2 (release)

While activating a new pod I got an error (lost the screenprint) just after filling the Pod, starting the pairing & priming sequence (see log at 00:34:11?
Fortunately, after clicking the retry button, Pod initialisation sequence was smooth and fully succesful.

Remark:
From the logs it looks like a comms failure. Can not explain as generally I have very steady RL/Pod connections with virtually no failures on the UI (other then by my own doing).
(FYI: going back in the log you will find lots of failures before starting a Pod de-activation: I walked downstairs without taking the phone :-)

[AndroidAPS.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4125764/AndroidAPS.zip)
[AndroidAPS._2020-01-28_00-00-00_.2.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4125765/AndroidAPS._2020-01-28_00-00-00_.2.zip)

bartsopers, 2020-02-17:
Thank for your report. This is normal behavior which can sometimes occur. If you ever experience a situation in which the retry button repeatedly doesn't work, which forces you to discard your pod, please report it under #60.

closed 2020-02-17 by bartsopers

vanelsberg, 2020-02-18:
Sure, Thanks.
In this particular case I think the driver handled the problem and was able to succesfully activate the Pod. Nice!