AAPS-Omnipod/AndroidAPS-OLD #82: Losing connections to RL and/or Pod

Kywalh, 2020-01-29:
**Version: 0.2**
**Test Description:** 
Phone and RL fully charged
Phone and RL rebooted
Pod already active
Pod/RL/Phone within 50cm over the complete night testing
Pump unreachable set to 60min

**Issue:** 
The RL or POD disconnects regularly (see screenshots) firstly noticed by looking at the basal graph color (dark blue is very very often a sign that the pump came back to reference basal).
No "Pump unreachable" notification over the night testing

[AndroidAPS._2020-01-29_00-00-00_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4128158/AndroidAPS._2020-01-29_00-00-00_.0.zip)
[AndroidAPS.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4128159/AndroidAPS.log)
[image]
[image]

eurenda, 2020-02-16:
Another example of pump unreachable
![content://com google android googlequicksearchbox CommonContentProvider/assist com google android apps gsa staticplugins assist screenshot ScreenshotProvider/com google android apps gsa staticplugins assist screenshot ScreenshotProvider/ScreenAssistScreenshots/Screenshot_20200216-081559](https://user-images.githubusercontent.com/1902827/74608513-79fa8880-5096-11ea-9761-b8b1c0315502.png)
With logs attached: #[Logs -pump unreachable - Feb 16 0815](https://drive.google.com/file/d/18wZeHtbR8k0HFuJODNgg5pftoLT79_7T/view?usp=drivesdk)

bartsopers, 2020-02-17:
The pump unreachable notification is a symptom of a communication problem. In the Pod History screen, you can (probably) see what the issue was. If you see repeated Communication Failed errors there, please upload logs and a screenshot of the Pod History screen to #92.

danielhers, 2020-03-26:
This happens to me a lot more with 0.3 than it used to with 0.2.
The workaround of killing AndroidAPS and then launching it again and switching profile always works, but it's annoying and also means I cannot trust the loop to keep me in range.
Latest time of event: 18:17
AndroidAPS 2.7-omnipod-0.3
Build: EROS_0.3-2020.03.14-21:01
Remote: git@github.com:danielhers/AndroidAPS.git
Flavor: fullrelease
Nightscout version: 13.0.2-dev
Engineering mode enabled 
[AndroidAPS_LOG_1585243126221.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4388716/AndroidAPS_LOG_1585243126221.log.zip)

bartsopers, 2020-08-18:
Closing this as we did a lot of improvements on both communication and local alerts. Also opened #150 for implementing an auto-reconnect mechanism for the RileyLink

closed 2020-08-18 by bartsopers