AAPS-Omnipod/AndroidAPS-OLD #84: Pump unreachable. Reset AAPS and RL and while Omnipod status would state connected there would be no communication with pod. Had to rest pod to remove as deactivation would not work

ohjay73, 2020-02-08:
[AndroidAPS_LOG_1581179817081.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4174665/AndroidAPS_LOG_1581179817081.log.zip)
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

ohjay73, 2020-02-08:
Issue started ~ 0920 MT logs attached

andyrozman, 2020-02-15:
No logs attached and it seems to be duplicate of #83

closed 2020-02-15 by andyrozman

ohjay73, 2020-02-15:
@andyrozman logs are in the first post AndroidAPS_LOG_1581179817081.log.zip
Reporting bugs