AAPS-Omnipod/AndroidAPS-OLD #86: Pump "Suspended" might be connected with Objectives

dar-kod, 2020-02-13:
AAPS Omnicore 0.2
Samsung Galaxy s7 edge
Android 8.0.0

It happened today early after midnight between 00:00 and about 01:00 o'clock. I attached 2 logs created at 00.00 and 1:00.

I got first sound signal and message "Pump unreachable" at 22:02 previous day and second at 00:35 this day. After that I tried to do everything what is suggested: toggle BT off and on, restart the phone, Omnipod refresh. In this time I have seen that "Pump unreachable" status. I repeated the process many times and it was not working. I had to go to Maintenance -> Import setting and I lost my pod.

First time I imported my settings, in objectives it shows that I didn't finish objectives 2 and 3. By all other objective it is written "CLEAR FINISHED" and below the date accomplished. I have used before Jeremy's version of AAPS and OmniCore driver and I saved those settings and used after installing AAPS for Omnipod eros 0.2. I suppose this problem might be repeated in the future. This happened with my second pod used with Omnipod 0.2.
[AndroidAPS._2020-02-12_00-00-00_.5.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4197329/AndroidAPS._2020-02-12_00-00-00_.5.zip)
[AndroidAPS._2020-02-13_00-51-28_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4197330/AndroidAPS._2020-02-13_00-51-28_.0.zip)

dar-kod, 2020-02-13:
For clarification, I started with "closed loop" in July 2019 and finished all objectives in October 2019.

The answer from objectives at androidaps dot org was classical and not helpful:

> please note that AAPS master does not support Omnipod yet. If you use different branches be aware of the risks and ask the developers of those branches for support.

>

andyrozman, 2020-02-14:
Did you restart RL too? Sometimes RL just stops working, light is on, but nobody is home...

closed 2020-02-15 by andyrozman

dar-kod, 2020-02-15:
@andyrozman, yes I switched also RL off and on.

Now I have the same issue again. I tried the following: Refresh button in the Omnipod tab, a profile switch, swithch RL off and on, restarting AAPS with Force Stop in the Settings -> Apps. In the Omnipod tab RL is connected, it is also changed temp basal from 0 to 1.95 but I cannot give bolus. But after short time RileyLink Status: **Not Started** -> RL switch off/on and it is **Bluetooth ready** but no connection.