AAPS-Omnipod/AndroidAPS-OLD #89: Pod Error 128 - Basal under infusion

marksieminski, 2020-02-15:
Issue happened with two days remaining on the pod.
Pod failure happened while sitting down watching tv. No bolus or SMB activities were occuring.  Blood sugar was stable.

Using V0.2

[AndroidAPS._2020-02-14_16-45-18_.20.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4208500/AndroidAPS._2020-02-14_16-45-18_.20.zip)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

bartsopers, 2020-02-21:
Discussed this with @itsmojo. This is probably a problem in our basal schedule encoding, which should be investigated further.

bartsopers, 2020-08-24:
Closing this as this issue is half a year old now and we haven't had any other reports of it. If you experience this again, please re-report and provide me with the exact basal schedule that you were using.

closed 2020-08-24 by bartsopers