AAPS-Omnipod/AndroidAPS-OLD #90: Event History Dissappears

marksieminski, 2020-02-15:
Event History appears to disappear on phone. Information such as IOB and temp basals are not loaded causing androidaps to think no IOB and dose more insulin.

Error seems to occur at any time with no specific events leading to AAPS to clear event history.

V0.2 is used.

[AndroidAPS._2020-02-14_16-45-18_.14.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4208506/AndroidAPS._2020-02-14_16-45-18_.14.zip)
[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

andyrozman, 2020-02-15:
This has nothing to do with driver... It might to do with your NS setup...

marksieminski, 2020-02-15:
Interesting.  I will review NS, but haven't had this issue before.  Understand this might be directly related to AAPS and NS rather than AAPS driver.

andyrozman, 2020-02-15:
It is not driver related... We just add events, we don't do anything with them...

marksieminski, 2020-02-15:
Interesting. All events are uploaded to nightscout and functioning correctly. Do you think that it could be a communications bug with AAPS?  I never had this issue on 0.1 (could be coincidence).  Regardless I redeployed nightscout to see if there is any correlation.

marksieminski, 2020-02-16:
Andy, you are correct in stating this is not an AAPS issue.  It was corrected by cleaning up nightscout database.  This can be closed.

closed 2020-02-16 by marksieminski