AAPS-Omnipod/AndroidAPS-OLD #91: Pump suspended - can not undo?

vanelsberg, 2020-02-15:
AndroidAPS 2.6-omnipod-0.2
Build EROS_0.2 (release)

Problem:
AAPS alerted on a profile change back to 100% (beep, mute, OK)
AAPS suspended Pump.
Pod status shows some _uncertain failure_s on "Get Pod Status".
Pod seems alive and is not beeping/screaming.

_Unable to resume pump from suspension._
Tried AAP exit -> for-kill through settings -> restart

Possible causes:
Unknown.

[AndroidAPS_LOG_1581722265917.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4208696/AndroidAPS_LOG_1581722265917.log.zip)

dar-kod, 2020-02-15:
AndroidAPS 2.6-omnipod-0.2
Build EROS_0.2 (release)

Problem: **Pump suspended** could not be fixed with working temporary fix:

1. Use the refresh button in the Omnipod tab
2. Do a profile switch

I tried also other recommended steps: 

1. Restarting AndroidAPS with Force Stop in the Settings -> Apps
2. Bluetooth off/on
3. RileyLink off/on

After those steps the RileyLink is connected but very short time like a few minutes and the pump is suspended. RileyLink battery is fully charged.

I also tried to switch to virtual pump and back to Omnipod but without result.

See the logs at time before 20:56 when I decided to change a pod.

[AndroidAPS._2020-02-15_20-40-10_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4209076/AndroidAPS._2020-02-15_20-40-10_.0.zip)

vanelsberg, 2020-02-16:
@bartsopers @andyrozman Might not be related?

_I am using an AAPS local profile_:
The initial view on the profile is always correct. But very often when opening the view for a 2nd time  profile scheduled times are shifted by 2 hours backwards (wrong timezone?) . A standard close/reopen AAPS solves this. Never saw this as much of a problem.
I have seen this behavior consistently since AAPS version 2.3

So general bug in AAPS, not related to the OmniPod driver. Still, thought it might be worth mentioning? Considering the fact that I suspected inconsistencies in message send to the Pod when switching profile that resulted in pump suspension as reported. Some sort of profile corruption that caused my problem maybe?

whynot2018, 2020-02-20:
I had this issue. If you have this issue you can't loop and give boluses.
so I need to change a new pod.
[AndroidAPS.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4230038/AndroidAPS.zip)

danielhers, 2020-02-24:
Just had this issue after a Pump Unreachable. 
Killed the app and bolused successfully but then it got suspended again. 
Profile switch to 101% for 10m fixed it.

Kaivish, 2020-02-24:
Anyone tried to just tap on refresh in pod tab? Had a pump suspended issue as pod was not contacted for more that 4 hrs. Consequently loop stopped. But i just went to pod tab and tapped refresh and pump came out of suspended state.
This suspended pump came up just after aaps restarted itself. Seeing aaps start logging on ns and shortly after occured the pump suspended issue. Maybe just a coincidence.

danielhers, 2020-02-24:
Yes, that's actually the first thing I tried, but it didn't help.

Kaivish, 2020-02-24:
Uploading logs for pump suspended issue. Timeline is 00.01.xx
[AndroidAPS_LOG_1582515940110.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4244162/AndroidAPS_LOG_1582515940110.log.zip)

danielhers, 2020-02-24:
Here are my logs. Pump unreachable at 00:08. Bolus at 03:08 and then profile switch.
[AndroidAPS._2020-02-24_03-05-46_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4244198/AndroidAPS._2020-02-24_03-05-46_.0.zip)

teleriddler, 2020-03-12:
I had the exact same problem today after weeks of stable 0.2 running.

Things I tried

andyrozman, 2020-03-26:
@vanelsberg I have seen this behavior consistently since AAPS version 2.3... Could you please report this to AAPS core please... (and check if similar issue was already updated)

elgu11, 2020-03-28:
I still have the same problem with Eros_03. @andyrozman 
[AndroidAPS_LOG_1584884289173.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4397952/AndroidAPS_LOG_1584884289173.log.zip)

vanelsberg, 2020-03-29:
> @vanelsberg I have seen this behavior consistently since AAPS version 2.3... Could you please report this to AAPS core please... (and check if similar issue was already opened)

Will do!
Just think this might also be related to driver logic in combination with comm errors during profile switch operations?

Observation:
Problem is always the result of a profile switch failure and can be solved by manually switching profile.

I would say switching profile suspends the pump. Then when comm errors occur on setting the new profile, the action is aborted and the pump is not resumable due to an "unknow" profile state?
Maybe, driver does/can not handle this state and/or AAPS does not get informed there is a problem?

elgu11, 2020-03-29:
@vanelsberg 
Of course, now I understand why I have started to get these errors that I did not previously with eros-01. From eros-02 I started with "Automation" and created a profile change at high glucose (> 9 mmol).

mick711, 2020-04-09:
Hi i have this issu today and send by email log, when it come sometime like this log when time and time zone issue come that solve the problem .

mistermintsgh, 2020-04-18:
I've just had this problem twice today.

Steps to recreate my instance:
1. Set profile change for set period (in my case I switched from profile A to profile B for 60 minutes)
2. At end of the predetermined profile switch duration alarms sound to say that there was a communication error and profile could not be switched back
3. Pump suspended message displayed on screen (see screen shot)
[image]

During this time my pump was still connected - I could refresh the status on the pod tab, could still bolus, and default basal was being delivered

Only way I could unsuspend the pump was to force a profile switch to another profile, then back to the one I wanted to be using.

avicorp, 2020-07-18:
Fix request: If the system is in a state of pump suspended due to a failing profile switch, massage will show "Suspended pump dou to comonication fail on profile switch. Check rileylink connectivity or try to reset the profile." Two buttons: "Ok" and "Reset profile".

bartsopers, 2020-08-18:
The false pump suspended notifications are fixed in 0.4

closed 2020-08-18 by bartsopers