AAPS-Omnipod/AndroidAPS-OLD #92: Multiple commands fail in a row due to 'Communication failed' error

bartsopers, 2020-02-17:
Sometimes, multiple commands fail in a row due to 'Communication failed' errors.

*NOTE: Please comment to this issue if you also experience this issue and please attach:*
- *Logs*
- *A screenshot of the Pod History screen* (you can find this on the Omnipod tab > Pod Mgmt > Pod History
- *Which steps you took to fix this issue* (resetting RL, restarting AAPS, turning off/on BT, rebooting phone?)

biwhite, 2020-02-17:
Experienced a pump suspended around midnight.  That eventually was rectified near 2am.
Trying to bolus later that day around 19:30, had numerous communication errors.  Again eventually rectified after a bunch of attempts.

Normally try AAPS restart, bluetooth toggles.  Reset RL configuration etc.
Logs for the past day attached, as well as a few screenshots from the failed bolus.
Screenshot from the overnight failure as well, showing when it started failing when you can see it went back onto default profile setting on the blue graph.

[image]

[image]
[image]
[image]
[AndroidAPS._2020-02-17_19-49-30_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4216288/AndroidAPS._2020-02-17_19-49-30_.0.zip)
[AndroidAPS._2020-02-16_12-58-58_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4216305/AndroidAPS._2020-02-16_12-58-58_.1.zip)
[AndroidAPS._2020-02-16_12-58-58_.2.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4216306/AndroidAPS._2020-02-16_12-58-58_.2.zip)
[AndroidAPS._2020-02-17_01-55-17_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4216307/AndroidAPS._2020-02-17_01-55-17_.0.zip)
[AndroidAPS._2020-02-17_01-55-17_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4216309/AndroidAPS._2020-02-17_01-55-17_.1.zip)
[AndroidAPS.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4216310/AndroidAPS.log)

biwhite, 2020-02-17:
Pod history screenshots, overnight pump unreachable stuff, and this evening communication failure.
[image]
[image]
[image]

bartsopers, 2020-02-22:
[AndroidAPS_LOG_1580726947481.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4239038/AndroidAPS_LOG_1580726947481.log.zip)
[image]

bartsopers, 2020-02-22:
@biwhite can you please add 1 screenshot and the corresponding log? Please refer to my example above.

avivaac, 2020-02-22:
[image]
[image]
[image]
[image]

avivaac, 2020-02-22:
[AndroidAPS._2020-02-22_07-49-00_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4239047/AndroidAPS._2020-02-22_07-49-00_.0.zip)

Kaivish, 2020-02-25:
Getting a lot of communication failed when getting pod status. But setting tbr is apparently working.
Sending all the logs when these issues happened. 
[image]
[image]
[image]
[image]
[AndroidAPS_LOG_1582660981238.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251964/AndroidAPS_LOG_1582660981238.log.zip)
[AndroidAPS._2020-02-25_13-14-24_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251965/AndroidAPS._2020-02-25_13-14-24_.0.zip)
[AndroidAPS_LOG_1582641359858.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251966/AndroidAPS_LOG_1582641359858.log.zip)
[AndroidAPS._2020-02-25_11-57-10_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251967/AndroidAPS._2020-02-25_11-57-10_.1.zip)
[AndroidAPS._2020-02-25_13-41-23_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251968/AndroidAPS._2020-02-25_13-41-23_.0.zip)
[AndroidAPS._2020-02-25_14-00-55_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251969/AndroidAPS._2020-02-25_14-00-55_.0.zip)
[AndroidAPS._2020-02-25_14-02-24_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251970/AndroidAPS._2020-02-25_14-02-24_.0.zip)
[AndroidAPS._2020-02-25_13-08-04_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251971/AndroidAPS._2020-02-25_13-08-04_.0.zip)
[AndroidAPS._2020-02-25_13-34-18_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251972/AndroidAPS._2020-02-25_13-34-18_.0.zip)
[AndroidAPS._2020-02-25_11-51-14_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251973/AndroidAPS._2020-02-25_11-51-14_.0.zip)
[AndroidAPS._2020-02-25_11-50-19_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4251974/AndroidAPS._2020-02-25_11-50-19_.0.zip)

vanelsberg, 2020-03-21:
Also seeing this on at least at two occasions.
First time I just ignored thinking I'd gone out of range. 

Second time at night with unexpected high BG when waking in the morning.
Surprisingly no alerts had been fired. Only noticed when trying to bolus which gave me the "communication failed/Mute/OK" pop-up.

Solved by toggling BT and force-kill/restarting AAPS

Unfortunately I lost the log's :-|
(Will monitor and post log's on when the problem reoccurs)

vanelsberg, 2020-03-29:
More of the same?
Log files export with 10 minutes after the incident:

[AndroidAPS_LOG_1585300225141.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4398366/AndroidAPS_LOG_1585300225141.log.zip)
[AndroidAPS_LOG_1585088381269.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4398367/AndroidAPS_LOG_1585088381269.log.zip)
[AndroidAPS_LOG_1585180912298.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4398368/AndroidAPS_LOG_1585180912298.log.zip)

mistermintsgh, 2020-05-09:
I'm having what I think is a very similar problem. Oddly, it's only really started to happen in the last couple of weeks (that I noticed).

If I don't manually hit refresh at least every 60-90 minutes the pod and RL will eventually lose connection. The status page will tell me that everything looks connected, but the pod history tells me communication has failed.

This is a particular problem over longer periods where I'm unable to interact with the phone (e.g. while sleeping) and I often wake up either high or low depending on what I've been doing if AAPS is unable to adjust my TBR.

Screen shot of last night's communication failure attached. Will also upload logs when I can get to a computer
[image]

mistermintsgh, 2020-05-14:
Logs attached for communication error overnight. Failure takes place around 0345hrs

[AndroidAPS._2020-05-14_00-00-13_.0[1].zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4627361/AndroidAPS._2020-05-14_00-00-13_.0.1.zip)

bartsopers, 2020-08-18:
Probably caused by a lost RL connection. Created #150 for implementing an auto-reconnect mechanism.

closed 2020-08-18 by bartsopers