AAPS-Omnipod/AndroidAPS-OLD #93: More times bolus attempt and screamer with the code number 049

whynot2018, 2020-02-20:
I adjusted a new pod and everything went as it should. I had to give bolus for my breakfast but it didn't occur. So I tried a few times or more and then I had this screamer error code 049.
[AndroidAPS.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4230077/AndroidAPS.zip)

bartsopers, 2020-02-21:
Can you please also provide the preceding log? This log only goes back until about 1 minute before executing the bolus, as you probably just restarted AAPS.

whynot2018, 2020-02-21:
[AndroidAPS._2020-02-19_10-35-27_.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4237993/AndroidAPS._2020-02-19_10-35-27_.zip)

bartsopers, 2020-08-18:
Closing this for now as this was probably in 0.1 or 0.2. We did a lot of improvements since then. Please open a new issue of you still experience this issue after the upcoming 0.4 release.

closed 2020-08-18 by bartsopers