AAPS-Omnipod/AndroidAPS-OLD #95: Safety issue : Basel rate not lowering even with reduced temp targets or profile

martianpc, 2020-02-24:
[image]

Verison .2 Basel rates are remaining at elevates levels despite efforts to lower them. Restarting seems to have no impact and neither does changing target or profile percentages. 

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

Kaivish, 2020-02-28:
Doing a profile switch won't reduce basal if aaps is dealing with a high bg. It will try to bring back bg to target. It will change isf and ic accordingly to profile switch. Base basal will be changed accordingly if bg is on target.
Its documented in the aaps docs. 
So this is not a bug or a safety issue.

andyrozman, 2020-02-29:
Profile Switches are actually the worst you can do, because once you do the profile switch, a lot of settings are reset. Closing, nothing to do with driver....

closed 2020-02-29 by andyrozman