AAPS-Omnipod/AndroidAPS-OLD #98: Aaps restarts by itself. Get a notification of no pod active and pump goes suspended state. Critical issue

Kaivish, 2020-02-24:
Something happened today at around pretty much same time. Around 23 is. Yesterday iy was logged at 22.57, today the same thing logged at 23.19
Aaps restarted by itself, don't know why. But start is logged in careportal, ns. When started again, it gives a notification of no active pod and it goes in suspended pump state. 
Could not get the logs in time yesterday, but today i managed to grab them. This seems to me  to be a critical issue. 
Attaching logs and screens. 

[AndroidAPS_LOG_1582515940110.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4247274/AndroidAPS_LOG_1582515940110.log.zip)

[image]
[image]
[image]
[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

Kaivish, 2020-02-26:
Same thing today again around same time
Timeline is 00.19
[AndroidAPS_LOG_1582680864281.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4253668/AndroidAPS_LOG_1582680864281.log.zip)

Kaivish, 2020-02-26:
Adding more logs
[AndroidAPS._2020-02-25_00-00-00_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257976/AndroidAPS._2020-02-25_00-00-00_.1.zip)
[AndroidAPS._2020-02-25_00-00-00_.2.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257977/AndroidAPS._2020-02-25_00-00-00_.2.zip)
[AndroidAPS._2020-02-25_00-00-00_.3.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257978/AndroidAPS._2020-02-25_00-00-00_.3.zip)
[AndroidAPS._2020-02-25_00-00-00_.4.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257979/AndroidAPS._2020-02-25_00-00-00_.4.zip)
[AndroidAPS._2020-02-25_00-00-00_.5.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257980/AndroidAPS._2020-02-25_00-00-00_.5.zip)
[AndroidAPS._2020-02-25_00-00-00_.6.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257981/AndroidAPS._2020-02-25_00-00-00_.6.zip)
[AndroidAPS._2020-02-25_00-00-00_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257982/AndroidAPS._2020-02-25_00-00-00_.0.zip)
[AndroidAPS._2020-02-24_23-19-02_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257983/AndroidAPS._2020-02-24_23-19-02_.1.zip)
[AndroidAPS._2020-02-24_23-19-02_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257984/AndroidAPS._2020-02-24_23-19-02_.0.zip)
[AndroidAPS._2020-02-25_21-01-57_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257985/AndroidAPS._2020-02-25_21-01-57_.0.zip)
[AndroidAPS._2020-02-25_21-01-57_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257986/AndroidAPS._2020-02-25_21-01-57_.1.zip)
[AndroidAPS._2020-02-25_21-01-57_.2.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257987/AndroidAPS._2020-02-25_21-01-57_.2.zip)
[AndroidAPS._2020-02-25_21-01-57_.3.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257988/AndroidAPS._2020-02-25_21-01-57_.3.zip)
[AndroidAPS._2020-02-26_00-36-08_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257989/AndroidAPS._2020-02-26_00-36-08_.0.zip)
[AndroidAPS._2020-02-26_00-36-08_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257990/AndroidAPS._2020-02-26_00-36-08_.1.zip)
[AndroidAPS._2020-02-26_00-36-08_.10.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257991/AndroidAPS._2020-02-26_00-36-08_.10.zip)
[AndroidAPS._2020-02-26_00-36-08_.11.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257992/AndroidAPS._2020-02-26_00-36-08_.11.zip)
[AndroidAPS._2020-02-26_00-36-08_.12.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257993/AndroidAPS._2020-02-26_00-36-08_.12.zip)
[AndroidAPS._2020-02-26_00-36-08_.2.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257994/AndroidAPS._2020-02-26_00-36-08_.2.zip)
[AndroidAPS._2020-02-26_00-36-08_.3.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257995/AndroidAPS._2020-02-26_00-36-08_.3.zip)
[AndroidAPS._2020-02-26_00-36-08_.4.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257996/AndroidAPS._2020-02-26_00-36-08_.4.zip)
[AndroidAPS._2020-02-26_00-36-08_.5.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257997/AndroidAPS._2020-02-26_00-36-08_.5.zip)
[AndroidAPS._2020-02-26_00-36-08_.6.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257998/AndroidAPS._2020-02-26_00-36-08_.6.zip)
[AndroidAPS._2020-02-26_00-36-08_.7.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4257999/AndroidAPS._2020-02-26_00-36-08_.7.zip)
[AndroidAPS._2020-02-26_00-36-08_.8.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4258001/AndroidAPS._2020-02-26_00-36-08_.8.zip)
[AndroidAPS._2020-02-26_00-36-08_.9.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4258002/AndroidAPS._2020-02-26_00-36-08_.9.zip)

Kaivish, 2020-03-01:
Still on ver0.2. Got the pump unreacheable issue and pump suspended after that.
Time line is 10.53 where pump unreacheable happened. 
[AndroidAPS_LOG_1583057510109.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4271630/AndroidAPS_LOG_1583057510109.log.zip)

andyrozman, 2020-03-12:
@Kaivish This is no longer relevant right? You found the problem.

Kaivish, 2020-03-12:
Well I thought so, but happened a couple of times again. Couldn't get the logs. Trying to figure out if its related to my phone

Kaivish, 2020-05-16:
Didn't see this behaviour for a couple of months now.

closed 2020-05-16 by Kaivish