AAPS-Omnipod/AndroidAPS-OLD #99: False Time/Time Zone change detected - pump suspended

mistermintsgh, 2020-02-28:
AAPS-Omnipod 0.2 detects a false time or a time zone change at midnight on my device. At that point connection to the pump is suspended, and has to be reconnected manually. 

No interactions are taking place leading up to this - it simply happens as the clock changes from 23:59 to 00:00hrs.

Logs attached leading up to 00:00 and from 00:00 to 08:31. Will endeavour to get more when this happens again.

Thanks

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

mistermintsgh, 2020-02-28:
[AndroidAPS._2020-02-28_00-42-08_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4268850/AndroidAPS._2020-02-28_00-42-08_.0.zip)
[AndroidAPS._2020-02-27_10-26-02_.2.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4268853/AndroidAPS._2020-02-27_10-26-02_.2.zip)

Kaivish, 2020-02-28:
Duplicate. Same as issue#55 opened earlier. Please upload your logs there and close this issue so there will be less duplicates.

closed 2020-03-02 by andyrozman