AAPS-Omnipod/AndroidAPS-OLD #100: Tbr is set to 0 by either internal driver or by user but in pod history when refreshed immediately after tbr is set, it shows other values

Kaivish, 2020-02-29:
Didnt get any logs yet, happed late at night. Posting pic of  pod history. I didnt listen to pod if it actually is 0.0u/hr or as what the pod history says. But swiping up and down in the pod history make the values/description at pod status change. Sometime 0.8, 0.1, 0.3u with duration 30 min. All that without changing the tbr ( which is still at 0.0u/hr according to home tab and pod tab) Not sure if basal is really suspended or just pod history is faulty. 

[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

andyrozman, 2020-03-02:
This is UI problem... Its fixced in 0.3

mistermintsgh, 2020-03-04:
I don't know if this is related, or if it's anything I should be worried about, (and I'll raise a separate issue if needed @andyrozman ) but like in the above screenshots I am getting lots of "Cancel TBR (Forced by user)" messages when I did no such thing.

There are plenty which say it was cancelled by the driver which makes me wonder whether it is a system issue

[image]

closed 2020-03-14 by andyrozman