AAPS-Omnipod/AndroidAPS-OLD #102: Current Pod address replaced by old pod address, unrecoverable

kborndorff, 2020-03-06:
[image]
[image]

Everything works fine then all of the sudden, profile is missing and when I activate profile again, the pop address was from a pod 6 days ago and the current pod is not able to be communicated with. Second time this happened in 2 days. 

Under the Pod window it says that the last communication was 6 days ago, but in reality it was only minutes before.

Only way to fix it was to reset the pod and start a new one.

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

kborndorff, 2020-03-06:
@
[AndroidAPS_LOG_1583471562997.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4296565/AndroidAPS_LOG_1583471562997.log.zip)

andyrozman, 2020-03-11:
Pod address has nothing to do with profile... 
Pod address is stored in Shared Preferences, and that can only be changed either by init commands, remove/reset commands, or by export settings or import settings.

vanelsberg, 2020-06-23:
I do not think this has anthing to do with profile.
Problem in triggered by restating AAPS.
And as reported, AAPS then continues with settings of the previous Pod.

I think the Pod is removed from settings on activation. Activated Pod adres is used (from memmory) but then reset on the first restart...

kborndorff, 2020-06-24:
This was my fault for importing settings, that seemed to have brought in the old Pod information. Wasn't aware that would cause the issue at the time, but good to know.

closed 2020-06-24 by kborndorff