AAPS-Omnipod/AndroidAPS-OLD #103: Cant see Omnipod as a pump option

reager, 2020-03-07:
I have cloned the repository (omnipod_eros branch) and built the project in Android Studio. When Im running it on my phone I do not see the omnipod option when Im supposed to select pump.

Build: EROS_0.2

biwhite, 2020-03-07:
Turn on engineering_mode

rajid, 2020-03-08:
I have followed instructions here:
https://readthedocs.org/projects/androidaps-altguide/downloads/pdf/latest/
and created the needed file "engineering_mode" under
Android->data->info.nightscout.androidaps->files
and have restarted the phone. 
Yet, it still doesn't show Omnipod as a pump choice.

When I built this version, I made sure my git branch was set to "omnipod_eros" under VCS->git branches.

closed 2020-03-11 by andyrozman