AAPS-Omnipod/AndroidAPS-OLD #104: Suspended pod in Open Loop-mode without error "No active pod"

Ljuslykta, 2020-03-12:
Twice, I've had the problem that without user input or indication from the app, the pod becomes suspended and a warning that there's no active pod. When looking in the RL tab, the "Last connected" is red, and when communication is refreshed the pump goes out of suspended and delivers insulin once again. 

The issues were discovered around the time I took screenshots, see the clock there for time.

Build version: EROS_0.2-2020.03.06-22:12
[AndroidAPS_LOG_1583936382490.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4325849/AndroidAPS_LOG_1583936382490.log.zip)
[AndroidAPS_LOG_1584005284203.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4325851/AndroidAPS_LOG_1584005284203.log.zip)

Ljuslykta, 2020-03-12:
[image]
[image]
[image]

bartsopers, 2020-08-18:
These issues have been fixed in the upcoming 0.4 release.

closed 2020-08-18 by bartsopers