AAPS-Omnipod/AndroidAPS-OLD #106: FEATURE REQUEST: Pod change blackout time

mistermintsgh, 2020-03-13:
This is an idea that has already been implemented in the Omnicore build; a user definable period in which a pod cannot be changed.

For example, I could set my blackout period to be 10pm until 7am, or whatever hours I choose. If my pod is set to expire during that blackout period, AAPS fires an alarm telling me I need to change it before the blackout period begins.

The idea is that if I happen to be working nights, or have an emergency change in the middle of the night, or any other reason why I am unable or want to change a pod at a generally inconvenient time that occurs regularly, I do not forget about it and am given ample warning to change before that period begins. If my pod expires outside of those hours, no special warnings are given.

andyrozman, 2020-03-14:
This seems like something that could be used in whole AAPS. Can you please open this same ticket on main project (and please attach sample pictures from omnicore build, just so that people have idea) .
We will leave this opened here, but this is not priority until everything else is done... In mean time maybe someone will implement it in AAPS Core, so we won't have to.

mistermintsgh, 2020-03-15:
Unfortunately I don't have a running build of Omnicore I can take screenshots from, and I don't plan on using it! Plus, it looks as though this feature was only enabled in the latest release.

Code is here, but I will also raise a general AAPS request for this feature.
https://github.com/winemug/AndroidAPS

bartsopers, 2020-08-28:
@mistermintsgh we're not planning to add this in the Omnipod driver. As Andy suggested, you could try to create this feature request in the main AAPS repo.

closed 2020-08-28 by bartsopers