AAPS-Omnipod/AndroidAPS-OLD #107: New AAPS-Omnipod "ACTION" tab does not allow resetting "Insuline age" and "Canula age"

vanelsberg, 2020-03-15:
AndroidAPS 2.7-omnipod-0.3-SNAPSHOT
Build EROS_0.2-341-g8c803f7e0

**Configuration:**
* Local AAPS profile

**Problem:**
New ACTION tab does not allow resetting "Insuline age" and "Canula age", only "Pump Battery Change". As a result AAPS main view (with option "Show status lights" enabled) shows wrong values.
See screenshot.

Additional remark:
I know the CAN/INS/BAT do not really aplye for OmniPod but for me the status lights do/could show valuable information on Sensor age, and Pod/Canula Age and Resevoir content.
So I do not want to disable the AAPS main view status line .
(AFAIK I can not disable showing "Insuline/Canula" values?)

*Expected behavior, either:*
- Must have:
Show additional buttons like "Insuline age" and "Canula age" for resetting
- Should have:
* Insuline- and Canula- age are automatically reset, f.i. on resetting "Insuline age".

*Feature request:*
* Auto reset values for Insuline, Canula and Battery on Pod init.

[image]

andyrozman, 2020-03-16:
Reopen this is Core product

teleriddler, 2020-03-16:
In AAPS core there is a the **Site Change** moved from the **Care Portal** to the **Actions tab** and is now under **Prime/Fill**.   

I have installed the latest version of AAPS 2.6 and can verify that this functionality exists there, however in AAPS-Omni, the icon has been removed, understandably as Omnipod really executes the Prime/Fill automatically in pod initialization.

https://androidaps.readthedocs.io/en/latest/CROWDIN/pt/Usage/CPbefore26.html

[image]

**Suggestion**

Would it be possible to enable the **Prime/Fill** button but just update the button text to **Change Site**?  If not, then we need a way to mark a site change so the Canula Age can be correctly reset and reflect site changes in NS.

andyrozman, 2020-03-16:
I will reset cannula age in code after Init is successful.

vanelsberg, 2020-03-16:
> I will reset cannula age in code after Init is successful.
That would fix for me!
Resseting "Insuline age" on init would also be logical for AAPS-Omnipod?

vanelsberg, 2020-04-16:
@andyrozman 
Thinking again on this I realize that resetting the _cannula age in code after Init is successful (milestone 6)_ is only solving part of the problem. As the screen print shows also "Insulin age" can not be reset through the action tab. Currently I am using "Pump battery change" to register Pod age but somehow this does not feel logical.
For OmniPod one would think there is no need to bother the user with manual interaction at all. 

_I would suggest also resetting "Insulin age" and "Pump battery change" on Pod init._
Also, as the "Pump Battery Change" button is then no longer needed it could be made invisible?

bartsopers, 2020-08-17:
@andyrozman could you look into @vanelsberg's suggestions?

vanelsberg, 2020-08-18:
@andyrozman @bartsopers I think this was solved with 0.4 dev.

bartsopers, 2020-08-18:
@vanelsberg Are you sure this is also fixed for isulin age and pump battery change?

vanelsberg, 2020-08-19:
AFAIK, yes.
Insulin age used to show cumulative time since last install.
From what I see on recent 0.4 dev commits, on the ACT tab "Insulin age" shows as "-".
Also on Nightscout, IAGE has no value.

[image]

vanelsberg, 2020-08-19:
Also, Canula age where perfectly reset on Pod change.

Pump battery rest needs manually issuing a "Pump Battery Change"  on the activity tab after Pod change.
Maybe this could also be automaticcally reset through Pod init?

Not yet tested on current dev 0.4 1 dev commit.
Will let you know for my next Pod change!

bartsopers, 2020-08-20:
@andyrozman could you please look at removing the "Pump Battery Change" button on the activity tab and resetting it automatically after activating a new pod instead?

vanelsberg, 2020-08-20:
> @andyrozman could you please look at removing the "Pump Battery Change" button on the activity tab and resetting it automatically after activating a new pod instead?

If that is feasable that would be awesome!

bartsopers, 2020-09-03:
Fixed:
- Reset Pump Battery and Insulin age after activating a new Pod
- Hide Pump Battery Change button in the Actions tab for Omnipod

closed 2020-09-03 by bartsopers