AAPS-Omnipod/AndroidAPS-OLD #111: Unable to communicate - pod is in wrong state

mistermintsgh, 2020-03-15:
Since update to 0.3 I have been unable to bolus. Following same steps as always, trying to bolus from either the calculator or the insulin tab/button along bottom of screen.

Error message shown below

[image]

Pod is connected as shown below:
[image]

Pod history shown below:
[image]

Logs to be attached shortly

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

mistermintsgh, 2020-03-15:
Most recent logs when this has occurred. Should've mentioned in the opening post, this was following a new pod change ahead of eating, and occurred around 19:15 to 19:20hrs

[AndroidAPS._2020-03-15_16-53-05_.0[1].zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4334987/AndroidAPS._2020-03-15_16-53-05_.0.1.zip)

andyrozman, 2020-03-18:
Problem happens when Initialization doesn't succeed and we are in in-between state. I added additional state, so when we are in this between state this is now displayed in tab, with state displayed, and system won't think that Pod is ready when it isn't.

mistermintsgh, 2020-03-19:
So how do I get from wrong state to right state?

The pod has already been initialised so as far as I can tell it can't be initialised again, and it is processing commands as basal instructions are being sent.

If pod enters this wrong state, is the correct thing to do just discard the pod and start with a new one?

mountrcg, 2020-04-19:
same thing happened to me an hour ago: https://discordapp.com/channels/629952586895851530/680407341820280863/701483178233167882

mountrcg, 2020-04-19:
I changed the pod. No issues deactivating the troubled pod. It also delivered standard basal rates but defied temp basal and bolus with comms errors.

vanelsberg, 2020-05-24:
See issue #126 
Is this the same problem?

larsvandoremalen, 2020-07-09:
I also have this problem with my current pump. Luckily running around very busy at the moment so basal only is actually working out fine at the moment :-) I am on 0.3 regular AAPS Omnipod branch. Will upload logs later today. If I can do anything else to help just let me know.  (@vanelsberg for me it looks very similar to your issue as well, I would say these are all the same).

teleriddler, 2020-08-07:
A lot of pod init issues were resolved in the latest dev 0.4.   You might want to try a quick build and see if this resolves your issue.  

**My current setup:**

**Version:** 0.4 (AndroidAPS 2.7-omnipod-04.-dagger-SNAPSHOT)
**Build:** EROS_0.2-978-gd446751db-2020.08.06-22:15) 

**Phone:**  Samsung Galaxy S7 Edge, Android 8.1 (Oreo)
**RileyLink** (Firmware BLE113: ble_rfspy 2.0, CC110: subg_rfspy 2.2)
**xDrip+ Version:** Nightly build
**CGM:** Libre 2 with LibreView patched app 2.3

bartsopers, 2020-08-09:
This has (probably) been fixed

closed 2020-08-09 by bartsopers