AAPS-Omnipod/AndroidAPS-OLD #112: Unable to Init Pod

bionictom, 2020-03-15:
After first installing AAPS Eros 0.3 I couldn't Init pod, the option was greyed out. I restarted AAPS but nothing changed.

Everything worked after I rebooted my phone.

The succesful priming was at 18:48, so the problem occurred before that. I attached the log.

[AndroidAPS_LOG_1584309245311.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4335214/AndroidAPS_LOG_1584309245311.log.zip)

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

teleriddler, 2020-08-07:
This may be linked to issue #125 and the latest dev 0.4 build has a lot of fixes that addressed and fixed that issue.

You might try building with dev to see if your issue has been resolved.

bartsopers, 2020-08-17:
Will be fixed in #134

closed 2020-08-17 by bartsopers