AAPS-Omnipod/AndroidAPS-OLD #113: Incosistencies between POD history and RL history

bionictom, 2020-03-15:
Nothing big, but I've noticed there's an inconsistency between the Pod and the RL history.

The Pod history is sorted by newest at the top, but RL is sorted by newest at the bottom.
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

teleriddler, 2020-03-17:
I would also like to see the RL history start with newest events at the top, a lot less scrolling during troubleshooting.

teleriddler, 2020-08-07:
I don't exactly remember where the RL history is but the POD History located in RL Stats on the POD tab is sorted by newest items first (at the top).  This may have been a UX change but I am not able to find the referred "RL History" in the latest dev 0.4.  I am guessing this was fixed or a UX change was made.   Can anyone explain where the "RL History" is located?

My current setup:

**Version:** 0.4 (AndroidAPS 2.7-omnipod-04.-dagger-SNAPSHOT)
**Build:** EROS_0.2-978-gd446751db-2020.08.06-22:15) 

**My Setup:**

**Phone:**  Samsung Galaxy S7 Edge, Android 8.1 (Oreo)
**RileyLink** (Firmware BLE113: ble_rfspy 2.0, CC110: subg_rfspy 2.2)
**xDrip+ Version:** Nightly build
**CGM:** Libre 2 with LibreView patched app 2.3

smuething, 2020-08-16:
This is resolved in the current dev version - both histories are sorted newest to oldest now

closed 2020-08-18 by bartsopers