AAPS-Omnipod/AndroidAPS-OLD #116: Lost connection with the pump, with AAPS unresponsive.

vanelsberg, 2020-03-25:
AndroidAPS 2.7-omnipod-0.3-SNAPSHOT
Build: EROS_0.2-341-g8c803f7e0-2020.03.14-23:06
Remote: https://github.com/AAPS-Omnipod/AndroidAPS.git
Flavor: fullrelease
Nightscout version: 0.12.3

*Configuration:*
Samsung S7, xDrip/FSL, Nightscout, AAPS local profile

*Problem:*
Lost connection with the pump.
From there, lock ups of APS, lost COB.
No alerts or notifications from AAPS
Needed force-shutdown through settings to solve and get AAPS going.

Note:
* Problem feels specific for v0.3 (did not experience any of this with v0.2)
* Looks similar to #82, but this issue does not mention AAPS becoming unresponsive.
* Have had at least one earlier occurrence of this problem a few days earlier

Time of event aprox before 2020-03-24 23:20

[AndroidAPS_LOG_1584647246465.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4380712/AndroidAPS_LOG_1584647246465.log.zip)

vanelsberg, 2020-03-31:
Similar behaviour?:

Pod Expired on 31.03.2020 10:17h, but expiry warning was ignored and alert silenced.

Tried de-activating Pod (already expired) at 10:58h - this ultimately failed: AAPS started getting totally unresponsive. In stead of a force-kill of AAPS I decided to wait it out: then after a few minutes AAPS finally crash/restarted (close/wait popup).
From there was well. Was able to Init a new Pod succesfully.

[AndroidAPS_LOG_1585567086372.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4411724/AndroidAPS_LOG_1585567086372.log.zip)

vanelsberg, 2020-04-15:
Similar behaviour?

Repro:

11:57h Bring RL Out of range (put it in a metal box)
11:58h Get Pod State (no response)
11:58h Try manual test bolus of 0.05u
AAPS locks up on the popup for acknowleging the bolus...
12:02h After some time restart AAPS on the "Close/Wait" popup.
12:03 Connecting at about 30 secs (??? RL is still out of range!)

AAPS reports is is "connected" (it cant be). Refresh button does not update or give an error.

[AndroidAPS_LOG_1586945151657.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4480543/AndroidAPS_LOG_1586945151657.log.zip)

bartsopers, 2020-04-19:
@vanelsberg do you mean that you get no alerts? or does it always crash?

vanelsberg, 2020-04-20:
1. It alkways crashes as described
From there (see note):
It recoveres when brogh in range
 or
It get in a unrecoverable state.

2. While the above is going, lots of errors reported,
But not a single alert

vanelsberg, 2020-04-20:
Another one:
Today I got out of range with Rl from the phoene for 45 minutes. Was alerted by xDrip because of high BG.

Lot of errirs in AAPS, no alert! 
Had to restart AAPS to get it communicating with RL

teleriddler, 2020-08-07:
I just tested this with DEV version 0.4 and after the manual test bolus out of range (in microwave) it continues to try to connect to the RL for up to 90 seconds and then I get the POD unreachable timeout (which is expected behavior).  After taking RL out of the microwave, reconnection is successful and the bolus in NOT delivered (also expected result).

My details on build and setup are listed below but I believe this has been resolved in dev 0.4 as AAPS did not crash and responded appropriately for with a timeout on bolus attempt during an active disconnection of the RL via BT and non delivery upon successful reconnection.

**Version:** 0.4 (AndroidAPS 2.7-omnipod-04.-dagger-SNAPSHOT)
**Build:** EROS_0.2-978-gd446751db-2020.08.06-22:15) 

**My Setup:**

**Phone:**  Samsung Galaxy S7 Edge, Android 8.1 (Oreo)
**RileyLink** (Firmware BLE113: ble_rfspy 2.0, CC110: subg_rfspy 2.2)
**xDrip+ Version:** Nightly build
**CGM:** Libre 2 with LibreView patched app 2.3

bartsopers, 2020-08-18:
Closing this for now as it does not seem reproducible. @vanelsberg please open a new issue if you still experience this after the upcoming 0.4 release.

closed 2020-08-18 by bartsopers

vanelsberg, 2020-08-19:
OK!
Will retest this scenario for dev/0.4.1