AAPS-Omnipod/AndroidAPS-OLD #117: Rileylink Unreachable

billymild, 2020-04-07:
I had an error message 2 times today that shows "RileyLink unreachable" on the Omnipod page. First time I turned off and turned on bluetooth. The second time the app locked up and I needed to force stop, clear cache and reopen. 

Build: EROS_0.3-2020.03.21-09:00

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

billymild, 2020-04-08:
Last night I had changed my pod. I didn't realize the RileyLink hadn't sent a message in almost 12 hours. I hit refresh and it came back no issue. Greenlight on RL was on the entire time. 

I just noticed in the last 20 minutes AAPS is showing RL as unreachable again. Logs attached. 
[AndroidAPS_LOG_1586360441130.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4451422/AndroidAPS_LOG_1586360441130.log.zip)

bartsopers, 2020-08-18:
Unfortunately, the connection with the RL sometimes fails. We haven't been able to solve that yet, and it's probably not possible to solve it for 100% as these issues can also be caused by the RL itself.
We did fix local alerts in the upcoming 0.4 release. You can enable these to get notified about communication problems.

closed 2020-08-18 by bartsopers