AAPS-Omnipod/AndroidAPS-OLD #118: ajust color threshold for res indicator

CommandMaster, 2020-04-08:
[image]

I've noticed that the res-indicator on my home screen is always yellow to warn me that the reservoir is low in my pod. Since the pod only gives accurate res reading after it is under 50units I would consider adjusting the threshold of the res indicator, so that it is yellow when there're only 50units in the res left.

haspden, 2020-04-16:
@CommandMaster that is configurable in your own plugin preferences...

Home Screen > Top Rigt 3 dots > Plugin Preferences > Advances Settings >
Now adjust your Threshold Warning Reservoir Level [U] to 49 or lower (depending on your preference)

[image]

haspden, 2020-04-16:
@andyrozman perhaps this issue should be closed now ?

Thanks as always

Henry

closed 2020-04-17 by CommandMaster