AAPS-Omnipod/AndroidAPS-OLD #119: Profile change during TBR causes undesired insulin delivery

djtremolo, 2020-04-15:
**Date**
15Apr2020, 10:41-10:50 (UTC+3 timezone)

**Version**
AndroidAPS 2.7-omnipod-0.3

**Expected behaviour**
User changes the profile
-> AAPS cancels the TBR before changing profile
-> new profile is loaded 
-> next Loop run detects that zero TBR is still needed
-> no basal insulin delivered after profile change

**Description**
When a profile change is made during a temporary zero basal set by AAPS, the zero basal is cancelled but not set again by AAPS. This is dangerous as it causes the pump to start delivering base basal rate without AAPS control/awareness.

**Reproducing**
Easiest way to reproduce is to use a fake bolus. 
_Note: The values in square brackets are the timestamp references in the attached log file._
1) preparation: make sure AAPS is not on zero TBR and it is running on 100% profile. [10:41:40.945]
2) use Insulin button to create a big-enough fake bolus with "Do not bolus, record only" option activated. [10:42:44.765]
3) run loop once via Loop tab to get the AAPS to start a zero TBR. [10:42:52.329]
4) make a profile change to 120% and wait until the change has been completed (10:44:22.235 -> 10:44:46.825)
5) verify that AAPS overview screen still shows zero TBR
6) listen to the Pod, you will hear it clicking (speed depends on your basal, assuming 1u/hr rate, it will click once in three minutes)
7) measure the time between clicks and calculate: basalRate = 180 / clickTimeInSeconds
8) reminder: don't forget to remove fake bolus in Treatments tab and cancel the 120% profile change

If you heard the pod clicking in step 6, you have confirmed the issue. If you take a look at the Loop diagnostic output, you'll see that the loop decided to continue zero TBR, but as there is no change (aaps thinks it already has zero tbr), no command is issued to pump driver.

**Log file**
[AndroidAPS_2020-04-15.txt](https://github.com/AAPS-Omnipod/AndroidAPS/files/4483324/AndroidAPS_2020-04-15.txt)

bartsopers, 2020-04-19:
Thank you for reporting this issue. I'll try to reproduce this next week

bartsopers, 2020-04-28:
Do you know if this issue also exists for other pumps?

kettmee, 2020-04-29:
I've been using only omnipod so don't know about other pumps. But I used omnicore before omniaaps and then there was no issue. Btw,noticed that this same issue occurs after a pod change and when a profile is activated without a time limit too.

Hanki Outlook for Android<https://aka.ms/ghei36>

djtremolo, 2020-04-29:
Similarly, I have used OmniPy and OmniCore, but none of the pumps supported by the official aaps.
I'll ask my friend who uses spirit combo and let you know if we get any results.

teleriddler, 2020-08-07:
I just tested this with the latest DEV build 0.4 and I did not get any clicks on my Omnipod during step 6.  Looks like the issue might be resolved in that version.

**Version:** 0.4 (AndroidAPS 2.7-omnipod-04.-dagger-SNAPSHOT)
**Build:** EROS_0.2-978-gd446751db-2020.08.06-22:15) 

**My Setup:**

**Phone:**  Samsung Galaxy S7 Edge, Android 8.1 (Oreo)
**RileyLink** (Firmware BLE113: ble_rfspy 2.0, CC110: subg_rfspy 2.2)
**xDrip+ Version:** Nightly build
**CGM:** Libre 2 with LibreView patched app 2.3

bartsopers, 2020-09-03:
@djtremolo I can confirm that this issue existed. It has now been fixed the omnipod_eros_dev branch.

closed 2020-09-03 by bartsopers

djtremolo, 2020-09-04:
> @djtremolo I can confirm that this issue existed. It has now been fixed the omnipod_eros_dev branch.

Excellent! Thank you!