AAPS-Omnipod/AndroidAPS-OLD #120: Screamer during SMB right after init

danielhers, 2020-04-19:
TIME OF EVENT HERE: 09:19
ISSUE DESCRIPTION: screamer a minute after initializing while delivering smb

AndroidAPS 2.7-omnipod-0.3
Build: EROS_0.3-2020.03.14-21:01
Remote: git@github.com:danielhers/AndroidAPS.git
Flavor: fullrelease
Nightscout version: 13.0.2-dev
Engineering mode enabled

danielhers, 2020-04-19:
[AndroidAPS_LOG_1587281353781.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4498687/AndroidAPS_LOG_1587281353781.log.zip)

bartsopers, 2020-04-19:
These logs don't contain any information about the Fault Event. Can you look for a log file that contains "PodFaultException"?

danielhers, 2020-04-20:
[AndroidAPS._2020-04-19_19-08-46_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4501445/AndroidAPS._2020-04-19_19-08-46_.0.zip)
Right, sorry. Here is the one after.

danielhers, 2020-04-20:
Looks like there was no "PodFaultException" in this event, but the pod started screaming and became unusable anyway. The log I sent does contain the event time. Possible related entries:

- 09:19:21.592: `isDeviceReachable. Response is invalid !`
- 09:19:25.055: `writeToData: RileyLink reports timeout`
- 09:19:31.69: `Caught exception in executeAndVerify. Verifying command by using cancel none command to verify nonce`
- 09:19:38.323: `Command unresolved (UNCERTAIN_FAILURE)`

After that I tried to deactivate the pod but that failed (09:21:35.611), so I had to reset pod (09:21:59.729) and then I could init and use another pod with no problem.

bartsopers, 2020-04-28:
Unfortunately, I can't understand what happened without the PodFaultException in the log, so I'm closing this issue. Thanks for reporting though!

closed 2020-04-28 by bartsopers