AAPS-Omnipod/AndroidAPS-OLD #121: During the process of starting a sensor in xDrip, AAPS Omnipod 0.3 disconnects with RL

dar-kod, 2020-04-23:
Build version: AndroidAPS 2.7-omnipod-0.3
Time: about 16:00
Date: April 20, 2020

Steps to reproduce a bug:

1. Open xDrip+ app.
2. Stop Dexcom G6 sensor and wait up the command is executed, usually during next reading. If the sensor is stopped, can be checked also in the log file. You may wait maximum 15 minutes.
3. Restart phone.
4. Start a sensor.

I started a sensor and opened AAPS to reconnect with RL. I think it was about 16:00. I tried for about next 1 hour to reconnect to RileyLink first restarting RL (of/on waiting for about 10 seconds between of and on), in Config Bilder -> Omnipod settings -> RileyLink Configuration searching for RL and connecting. But it was not successful. I tried to apply  different work around solutions: force stop AAPS, BT of/on... After repeating and trying many times, after maybe 1 hour or more, it started to work again.

Here are 2 log files. I think that AAPS Omnipod connection restored in second file.
[AndroidAPS._2020-03-20_00-00-02_.2.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4525734/AndroidAPS._2020-03-20_00-00-02_.2.zip)
[AndroidAPS._2020-03-20_00-00-02_.3.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4525735/AndroidAPS._2020-03-20_00-00-02_.3.zip)

andyrozman, 2020-06-26:
Did you try to restart RL too?

dar-kod, 2020-06-27:
Yes, I do this also, if such problem appear, but I do not remember when exactly I did it.

bartsopers, 2020-08-18:
Unfortunately I cannot pinpoint this to a problem in our code. If these issues still exist after the upcoming 0.4 release, please report again.

closed 2020-08-18 by bartsopers