AAPS-Omnipod/AndroidAPS-OLD #122: Not receiving bolus

reager, 2020-04-25:
[AndroidAPS.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4533576/AndroidAPS.log)

Build version: AndroidAPS 2.7-omnipod-0.3
Time: about around 18:10
Date: April 24, 2020

Multiple times since updating to 0.3 I have been experiencing that I not received the calculated bolus without any indication that there would be any issues.

Steps:
1. Use the calculator
2. Submit bolus
3. Progress bar increases and pod is ticking
4. Every now and then when the progress bar goes to 100 and then disappears and my pump ticks as it usually does when delivering insulin I do not actually get any insulin. In the graphs there are no indication of insulin delivered, the same when looking at treatment. But if I go to the pod tab it says that I got the bolus.

This causes severe issues for me. 

Any ideas?

bartsopers, 2020-04-28:
I don't really understand what you're saying. If the pod is ticking, it is delivering insulin, so you got your bolus.

reager, 2020-04-30:
The issue is that I do not get any insulin. 

Everything looks like the pod is dispensing insulin, and the Pod tab in the App have registered the bolus, but there is nothing in the graphs or treatment tab (and IOB does not increase), and from my blood sugar I can see that I did not get any insulin.

This happens multiple times every day, especially when using the calculator. When using the Insulin button it seems to work.

teleriddler, 2020-08-07:
After reading through the posts above I would like to ask where you are looking to verify your IOB?

Green IOB indicator for each entry in the Treatments->Bolus tab?
Main IOB indicator on the Home Screen?
IOB from the IOB data section of the SMB tab?

Also, keep in mind that when a temporary basal is set to zero (due to too much bolus) that a negative IOB is registered and added to the equations that AAPS Loop  uses to calculation the active IOB.  If there has been a lot of zero temping (temporary basal of 0 units) then a large negative IOB can build up and this is subtracted from the Bolus IOB given which can display a much smaller actual IOB.

Just wondering if this is the case for you?

bartsopers, 2020-08-18:
I cannot reproduce this issue so I'm closing it. Please report if you experience this issue again, and make sure to attach logs and screenshots.

closed 2020-08-18 by bartsopers