AAPS-Omnipod/AndroidAPS-OLD #123: Bolus started but missing

illanz, 2020-04-25:
AAPS omnipod 2.7 - 0.3
I've launched a bolus for 90gr of carb / 10 units through the calculator. The orange progress bar has started, here are the pictures 20 minutes later : zero active insulin and zero carb from the bolus.
Now I'm unable to know if I had my insulin or not (I ate a pizza...)

Log is available (very long - should I post it here?) 
[image]
[image]
[image]
[image]

bartsopers, 2020-04-28:
Can you please attach the log in a file? And do you remember at what time the bolus started?

illanz, 2020-04-29:
Bolus was started at +/- 12:15 or 12:20.
Here is the full log

illanz, 2020-04-29:
[AndroidAPS.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4553760/AndroidAPS.log)

andyrozman, 2020-05-07:
This could also be Wear issue...

illanz, 2020-05-07:
🤔

bartsopers, 2020-08-18:
I looked at you log. I found very little communication with the Pod in the log, actually only 1 TBR command. As there's also no reproduction steps, there's not much I can do, so I'm closing this issue. If this happens again, please open a new issue and make sure to include the correct logs.

closed 2020-08-18 by bartsopers