AAPS-Omnipod/AndroidAPS-OLD #124: Feature request to get TDD button in action tab

Kaivish, 2020-05-09:
[image]
Would be useful to be able to get this button for stats purposes. 

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

mountrcg, 2020-05-11:
TDD is part of the statistics menu item of the upper right menu.

andyrozman, 2020-06-26:
TDD with Pod is not possible, because Pod has no history... This is information that is retrieved from history... If we enabled TDD it wouldn't work for Omnipod... I am closing this ticket.

But like @mountrcg said, TDD can be viewed in Statistics...

closed 2020-06-26 by andyrozman