AAPS-Omnipod/AndroidAPS-OLD #125: ommunication timeout attempting to initialize pod

eurenda, 2020-05-13:
Reporting bugs
--------------
- **22:50 PDT - Freshly compiled omnipod-eros-dev branch with engineering_mode enabled, rebooted Samsung Galaxy S9 with Android 10, scanned for RileLink which showed "Connected" under OMNIPOD menu, proceeded to POD MGMT and selected Init Pod where I received _"connection timeout"_ .  Attempted several phone reboots, BT enable/disable, and AAPS exit/kill without successfully able to initialize pod (attempted with a new 2nd pod and obtained same result).
  the problem
-BUILD: AndroidAPS 2.7-omnipod-0.4-dagger-SNAPSHOT
Build: NoGitSystemAvailable-2020.05.11-21:20
Remote: NoGitSystemAvailable
Flavor: fullrelease
Nightscout version: 13.0.1
Engineering mode enabled 

See attached logs and AndoirAPS preferences
[AndroidAPS_LOG_1589349001812.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4624859/AndroidAPS_LOG_1589349001812.log.zip)

tgruszka, 2020-05-13:
Same issues as above. System was communicating fine until 20:30, then I started experiencing the same issues as eurenda listed above. 

[AndroidAPS_LOG_1589334412314.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4624962/AndroidAPS_LOG_1589334412314.log.zip)

Kaivish, 2020-05-22:
Same issue here on dev 0.4. It takes for ever to initialize the rl. A few exits and restarts of aaps eventually gets rl going but pod can't be initialized
[image]
[image]
[AndroidAPS_LOG_1590182333254.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4670456/AndroidAPS_LOG_1590182333254.log.zip)

Time is around 22.50~23.00

bartsopers, 2020-08-06:
I also experienced this issue, but it magically disappeared for me.... Can you try and see if you still experience it? @eurenda @Kaivish @tgruszka 
And if you still experience it, can you please provide logs?

teleriddler, 2020-08-07:
Just tested this on the latest DEV build (0.4 AndroidAPS 2.7-omnipod-04.-dagger-SNAPSHOT, Build: EROS_0.2-978-gd446751db-2020.08.06-22:15) and pods now successfully initialize.

Many others in the Discord chat have built and tested as of last night and reported being able to initialize pods as well.

My Setup:

Samsung Galaxy S7 Edge, Android 8.1 (Oreo)
RileyLink (Firmware BLE113: ble_rfspy 2.0, CC110: subg_rfspy 2.2)
xDrip+ Nightly build
Libre 2 with LibreView patched app 2.3

bartsopers, 2020-08-09:
This has been fixed today.

closed 2020-08-09 by bartsopers

eurenda, 2020-08-09:
> I also experienced this issue, but it magically disappeared for me.... Can you try and see if you still experience it? @eurenda @Kaivish @tgruszka
> And if you still experience it, can you please provide logs?
> Also any additional information about your specific situation might help. For example, I can reliably reproduce this issue by switching between RileyLinks

I just compiled this 

> I also experienced this issue, but it magically disappeared for me.... Can you try and see if you still experience it? @eurenda @Kaivish @tgruszka
> And if you still experience it, can you please provide logs?
> Also any additional information about your specific situation might help. For example, I can reliably reproduce this issue by switching between RileyLinks

@bartsopers I still get the issue with communication timeouts on Galaxy S9 with RL, see attached logs from compile from from 90 minutes ago on the dev branch

bartsopers, 2020-08-10:
@eurenda in the log you provided, I see a log line that I changed in the latest version which contains my communication fixes, so apparently you're not running the latest version. Please make sure that you run the latest version and try again.