AAPS-Omnipod/AndroidAPS-OLD #126: Upgard

vanelsberg, 2020-05-23:
AndroidAPS 2.7-omnipod-0.4-dagger-SNAPSHOT
BuildVersion: EROS_0.2-973-g19675b195-2020.05.23-11:33

_Configuration:_
Samsung S7, xDrip+

vanelsberg, 2020-05-23:
### Tested activating a new Pod. ###

Not surprisingly de-activating the current active Pod resulted in a "Communication failed: the Pod is in a wrong state" error.

The _"Reset Pod"_ option took care of that.
From there activating a new Pod was as usual - no problems.
RL Comms also seem to be stable as usual.

### Logfiles ###
Logfile as created after installation, Resetting the active Pod and activating a new Pod:

[AndroidAPS.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4672537/AndroidAPS.log)

### Screen shots ###
[image]

[image]

[image]

[image]

[image]

vanelsberg, 2020-05-23:
## Problem not solved ##

Lost connection to RL while going for coffee, leaving RL on my Desk....

RL reconnected without problems, including Pod serial, but n ow communication to the Pod is back to _"the Pod is in a wrong state"._ So this is the exact same situation I had when trying to take over an active Pod when upgrading to 0.4....

_I would say there is definitely a problem met omnipod_eros_dev_

Will downgrade to V0.3 after adding the logs below

### Logfiles ####

[AndroidAPS.log](https://github.com/AAPS-Omnipod/AndroidAPS/files/4672605/AndroidAPS.log)

[AndroidAPS._2020-05-23_23-56-22_.1.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4672604/AndroidAPS._2020-05-23_23-56-22_.1.zip)

[AndroidAPS._2020-05-23_23-56-22_.0.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4672606/AndroidAPS._2020-05-23_23-56-22_.0.zip)

vanelsberg, 2020-05-24:
Uninstalled 0.4 and installed 0.3
Imported v0.3 settings.

Surprise: AAPS started communicating with POD2 (activated with 0.4).
How is that possible as the POD in settings was de-activated/expired? 
AAPS also starts complaining about the POD2 being expired and after some time errors out on POD is expired responses.

vanelsberg, 2020-05-24:
Do not get it, does not make sense. In a nutshell, what is looks like (just guessing).
I would say V0.4 messes up with the Pod address.
Programming and initially successfully using the  Pod address of the previous Pod address (but storing a the new). On restart it switches to the new pod address which is not the address that was programmed to the Pod?

### Observations: ###

1. Upgrading form v0.3 to V0.4: fails (is AAPS somehow using the wrong Pod address?)

2. On V0.4 the Pod is reset. Activate new Pod initially is successful but AAPS looses the Pod address on app restart. Does it maybe program and initially use the old Pod address of the de-activated Pod (see2)? Then on restart, the new Pod address stored obviously does not work on the new Pod?

3. Downgrading to original V0.3 still has the old Pod Address of the expired Pod in settings,
However it can communicate with the newly activated Pod from v0.4?

vanelsberg, 2020-05-24:
See also #111

vanelsberg, 2020-05-28:
Additional comment:
This report , quite unexpectedly became a long story.
I guess it has to much information at once.

### In short: Repro steps: ###

1. While running an active Pod: export settings, uninstall V0.3 and install V0.4, import settings.
2. Deinit the active Pod (which is erroring "wrong state"). Activation should be successful.
3. Activate new Pod. PodInit should be successful, Pod is communication and OK.
4. Restart AAPS. Pod starts erroring "wrong state". Unrepairable....

5. Uninstall V0.4 and reinstall V0.3, import original V0.3 settings.
6 AAPS is communicating with active Pod (new activated on 0.4)

In my case the previous Pod from V0.3 was about to expire, so after initially communication with the Pod AAPS started complaining about the Pod being expired, then after a few minutes stopped working all together. Pod lost...

From there nothing wrong with the system for V03:
Activated a new Pod that worked for the full 80 hrs without any problems.
As usual, no comm errors.

vanelsberg, 2020-05-28:
I would suggest 2 tests:

1. While on an active Pod: 
Export settings. Uninstall V0.3, Install V0.4, import settings.

2. Make sure there is no Pod activated on V0.3 (When necessary deactivate Current Pod first)
Then export settings, uninstall V0.3, Install V0.4, import settings.

vanelsberg, 2020-05-30:
**Executed test case #2.**
Same behavior: after initially activating the Pod successfully, lost communication after restarting AAPS, erroring "the Pod is in a wrong state".

_Tried saving the newly activated Pod by capturing it with v0.3. That failed (with the same error) making me think the Pod address as exported by v0.4 json settings does not correspond the the actual Pod address?

(As the Pod was activated ok and responsive up until AAPSv0.4 restarted I find it unlikely  there is anything wrong with the Pod state...)_

vanelsberg, 2020-06-19:
Looks related to issue #102?

bartsopers, 2020-08-06:
Fixed :-)

bartsopers, 2020-08-06:
Thank you for the extensive research & reproduction path @vanelsberg!

closed 2020-08-06 by bartsopers

vanelsberg, 2020-08-08:
Top! Thanks for solving this Bart! 
Will start testing this a.s.a.p.