AAPS-Omnipod/AndroidAPS-OLD #127: Pressing cancel bolus button does not cancel the bolus

mistermintsgh, 2020-06-07:
At 7.37 this morning I was woken by a hypo alarm and went to treat it with Jelly Babies. By force of habit and being slightly confused because of the hypo, I went to the calculator and entered the carb amount.

The calculator suggested a bolus of 0.8U and without thinking I pressed OK. Quickly realised what I'd done when the "waiting for pump" dialog appeared so pressed cancel. It told me it had cancelled the bolus delivery, then went ahead and did it anyway.

See screenshot on pod history below.
[image]
[image]

Log to be attached as soon as I can get to a computer

Android APS 2.7 Omnipod 0.3
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://github.com/MilosKozak/AndroidAPS/wiki/Accessing-logfiles
- Open an issue at https://github.com/MilosKozak/AndroidAPS/issues/new

mistermintsgh, 2020-06-08:
Logs attached

[AndroidAPS._2020-06-07_00-00-02_.1[1].zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/4744947/AndroidAPS._2020-06-07_00-00-02_.1.1.zip)

Ljuslykta, 2020-06-23:
Got the same problem today. Tried to cancel a bolus right after pressing deliver, and it still both visually progressed and I heard the clicks.

levikemp, 2020-07-23:
@mistermintsgh looking at the log file and the image above can I ask if the time stamps I'm seeing are correct? In regards to the cancel being logged prior to the bolus.

mistermintsgh, 2020-07-23:
@levikemp Yes, should be. But admittedly the image makes it look like I cancelled the bolus before I started delivering it! It was literally just a second or 2 after pressing the deliver button that I pressed the cancel button.

bartsopers, 2020-08-18:
I'm sorry for the late reply, this issue somehow escaped my attention.
This is reproducable by immediately cancelling the bolus (before the Pod actually started delivering the bolus). I'll look into this ASAP

bartsopers, 2020-08-19:
This has been fixed in a branch that will shortly be merged to omnipod_eros_dev.

closed 2020-08-19 by bartsopers