AAPS-Omnipod/AndroidAPS-OLD #138: OpenAPS SMB does not take Max IOB setting from preferences

vanelsberg, 2020-08-16:
## Dev/0.4, OpenAPS SMB:

### Problem:

OpenAPS SMB Seems to ignore the Max IOB setting in preferences.
Under "constraints" it says "Safety: Limitting IOB to 3.0 U because of max value in preferences". 
Also "Profile:" lists "max_iob: 3.

#### Observations:
3.0 U seems fixed?
No matter what value I set as "Maximum basal IOB OpenAPS can deliver".
When selecting OpenAPS, "Profile:" does list the correct setting.

vanelsberg, 2020-08-16:
Any relation to current branch/merge and/or EROS driver?
Or is maybe this AAPS related...

vanelsberg, 2020-08-16:
[AndroidAPS_LOG_1597576169957.log.zip](https://github.com/AAPS-Omnipod/AndroidAPS/files/5080543/AndroidAPS_LOG_1597576169957.log.zip)
[image]
[image]

bartsopers, 2020-08-17:
I think this is a bug in the upstream dev version that we were using for 0.4. Closing for now, please re-report if this issue still exists after the 0.4 release.

closed 2020-08-17 by bartsopers

vanelsberg, 2020-08-18:
Yes..
Will re-report when still exists.

vanelsberg, 2020-08-20:
Problem seems to be solved for AAPS 2.7-omnipod-0.4.1-SNAPSHOT